openvehicles/Open-Vehicle-Android #38: Implement a feature to calculate estimated charge done time for other cars than Twizy

jilleberends, 2015-03-02:
The Twizy features a charge done estimate since the last version. It would be a killer feature to have this available for other cars as well, as this data is currently only available by sending an SMS.

dexterbg, 2015-03-02:
In fact the way it's done now for the Twizy is not very good and should not be adopted, as it needs separate msg commands to query the ETR data.

Mark already included most of the necessary fields in the STAT msg, so I can switch to using them after updating the Twizy firmware to fill them -- which is mostly a ROM space issue. Only the Roadster firmware currently fills the new STAT fields.

dexterbg, 2015-03-22:
I extended the STAT msg to include the ETR data as needed and switched Twizy firmware & App over to use these fields. So framework 2.7.2 + App 3.8.2 now enable any vehicle implementation to use this feature, they only need to fill the virtual car variables.

closed 2015-03-22 by dexterbg