openvehicles/Open-Vehicle-Android #40: Charge settings dialog hangs

gluca, 2015-03-02:
Scenario: Android app 3.6.0 on Motorola Moto G (1st generation) running Android 4.4.4.

Tapping on battery icon opens the new charge settings dialog. Buttons are actives but Both cursors "Sufficient SOC" and "Sufficient Range" are disabled and a white circle of death (:-)) stay on top of the dialog.

Ok/Cancel buttons work (both closing the dialog).
Cursors are stuck on 0.
Twizy is disconnected from OVMS box (server has no data since 29 days).

I've installed the 3.6.0 from the link provided by Michael in the ovmsdev list. I've removed previous version (the one actually available on Google Play App) before install 3.6.0.

dexterbg, 2015-03-02:
The settings dialog needs to communicate with the module to query and set the alerts. So this will only work if the module is online.

closed 2015-03-02 by dexterbg

gluca, 2015-03-02:
Maybe a dialog after a reasonable timeout (10seconds) would be helpful.

dexterbg, 2015-03-02:
I've included an informative label about what's going on, that should do. Timeouts for sendCommand need to be very high, the user normally cancels manually before.