openvehicles/Open-Vehicle-Android #43: Material design

jilleberends, 2015-03-22:
Open Vehicles version number 4 is getting closer... how about a redesign, Material design style? 
Here's a quick mock-up I made of the first screen:

[image]

dexterbg, 2015-03-23:
Nice. I'd prefer something more technical looking though, this is more like a business presentation.
I like the BMW i3 App design:
http://www.autovolt-magazine.com/wp-content/uploads/2013/07/bmw-i-app.jpg

jilleberends, 2015-03-23:
Wow, I actually like that a lot!
Now I think of it... the current OVMS resembles the Model S interface:

http://www.tech5.nl/files/2013/09/tesla-modle-s-app.jpg

http://insideevs.com/wp-content/uploads/2013/12/model-s-app.jpg

Currently the OVMS Battery screen is pretty cluttered: Car image, connector, battery, range, eta, and not everything is alignedor has the same style. I could start by using the Model S app as an inspiration, e.g. on the battery bar, and improve the quality of the connector graphics, gather the data in a somewhat nicer way. What do you think?

markwj, 2015-03-24:
Well, strictly speaking OVMS came first so the Model S interfaces resembles ours ;-)

jilleberends, 2015-03-24:
fair point :)
 +1

jilleberends, 2015-03-26:
For starters, I did a slight redesign of the Charger plug graphic. It should look a bit better on bigger screens now.

[image]

dexterbg, 2015-03-27:
Looks a lot better. Can you rework the charger_button.png as well? The low res slider looks strange in the new plug.

jilleberends, 2015-03-27:
Thanks.
Working on it :-)

jilleberends, 2015-03-28:
Something like this?

[image]

dexterbg, 2015-03-29:
Looks good, but the arrows lack contrast, especially on low resolutions. Maybe adopt the red/green marker style, or fill the arrows otherwise?

jilleberends, 2015-03-30:
[image]

jilleberends, 2015-04-02:
I'm playing around with GUI-elements to refresh the lower part of the Battery screen a little more. Here´s a rough idea of what I´m having right now, this is far from final in any way, but feel free to shoot your ideas
Main goals I'm going for:
- user friendly
- clarity
- easy on the eyes
- following current app design standards
- no loss of data/features from current screen
[image]

Saftwerk, 2018-01-07:
For my old eyes, the little text is too small to read without glasses.
Please make the smallest font bigger.
[image]
[image]

glynhudson, 2018-08-01:
Wow! I love the material design mockup. Please keep the dark theme, like the Tesla app I think a dark theme looks good. 

+1 for increasing the font size and button size for the important metrics and operations.

amralomari, 2019-05-07:
Thanks for the great work :)

I think there is a bug; when you arrive at Location screen you can't swipe back to the Car screen, the only way is to use the drop-down list.

Saftwerk, 2019-05-07:
You must start swiping from the edge of the screen. Don't start swiping from the middle area.Kind regards-------- Originalnachricht --------Betreff: Re: [openvehicles/Open-Vehicle-Android] Interface redesign (#43)Von: amralomari An: openvehicles/Open-Vehicle-Android Cc: Saftwerk ,Comment Thanks for the great work :)
I think there is a bug; when you arrive at Location screen you can't swipe back to the Car screen, the only way is to use the drop-down list.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/openvehicles/Open-Vehicle-Android/issues/43#issuecomment-489987814",
"url": "https://github.com/openvehicles/Open-Vehicle-Android/issues/43#issuecomment-489987814",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]