openvehicles/Open-Vehicle-Android #59: A/C Control button missing?

glynhudson, 2018-08-01:
On some app screenshots, I've seen an A/C control button on the Car page. This seems to be currently missing from the app (at least on my device). It took me a while to figure our that setting the HomeLink option to 1 turned on the AC and 2 switches it off. Have a missed something? 

Keep up the good work. Super impressed with my first 12hrs using OVMS 👍 

[image]

dexterbg, 2018-08-01:
The A/C button is shown depending on the car capabilities reported. The capabilities are a V2 feature that has to be ported to V3 yet. There is an open discussion on this on the mailing list, see:
http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/005344.html

glynhudson, 2018-08-16:
Thanks for explanation. How is the porting progress progressing? I've joined the mailing list but struggling to follow the messages. Have you thought about moving the discussion to GitHub or a dedicated form? I run a discourse from for openenergymonitor. This works well, happy to help set one up http://community.openenergymonitor.org

dexterbg, 2018-08-19:
The mailing list archive is helpful on catching up:
http://lists.openvehicles.com/pipermail/ovmsdev/
It's also searchable with Google using the "site" operator.

Feel free to join the discussion and/or define github issues. There are no plans to move the list to a forum.

glynhudson, 2018-11-13:
It looks like this PR is added climate control to the V3 framework: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/168

DTFUHF, 2019-05-05:
I'm running the 3.2.001 release of OVMS that includes this change, however I don't see the climate button(s) in the app. Is there another step to take to configure this?

DTFUHF, 2019-05-05:
I'm running the 3.2.001 firmware, but the AC button doesn't appear in my app. Is there another config step that needs to be done?

dexterbg, 2019-05-06:
As there has been no progress on the capabilities support in V3, I have now added the A/C button for every Nissan Leaf vehicle in the App. This change is included in release 3.12.10, which is now processed by Google.

In effect we do not use the capabilities for anything now, so we're free to define a completely new scheme as necessary.

closed 2019-05-06 by dexterbg

glynhudson, 2019-05-06:
Awesome! Nice work, just tested works great 👍