openvehicles/Open-Vehicle-Android #60: Display 12V aux battery voltage on the Car page?

glynhudson, 2018-08-01:
Has it been discussed to show the 12V aux battery voltage on the Car page? If you think this is a good idea I will try and make it happen! 

Thanks.

dexterbg, 2018-08-01:
The 12V voltage is already shown on the car page, but for the Twizy only, as the Twizy has a tiny 12V battery. Up to now you're the first to ask for that info on another car.

The Twizy display reuses the ambient temperature space, as that's not available on the Twizy:

[image]

If you're going to add the 12V info as a standard display, you need to place it somewhere else. Btw, changing the layout of the page is also an open todo… ;-)

The 12V info is available on the vehicle info page (see menu on car page) for all vehicles.

glynhudson, 2018-08-02:
12V battery voltage is very important for Nissan leaf since it's well known that the leaf is poor at keeping it's 12V topped up and a known common point of failure. It's good that the 12V voltage is accessible from the menu page, I had not noticed that until now.

Wow, 50763km! That a lot of km for a Twizzy :-p

dexterbg, 2018-08-02:
Not at all, it's about average for a six year old Twizy. High sixth year Twizy mileages are around 150.000 km, those normally are on the second battery as the 75% SOH warranty is reached at around 100.000 km (~1.500 full cycles).

glynhudson, 2018-08-02:
Impressive, I've driven a Twizzy a number of times and really wanted one but got put off by the battery lease. If it's possible to get one battery owned I would get one.

ltjessem, 2019-10-15:
There's lot of unused stuff on the leaf car page as well, why not add 12V?

glynhudson, 2019-10-15:
Agree, I still think it's a good idea to display the 12V voltage for all cars. The 12V battery voltage is an important metric to keep an eye on. A healthy 12V is critical to the reliability of all EV's e.g the LEAF is particularly bad at sufficiently charging the 12V. Usually the only indication that the 12V is running low is lots of strange error messages.