openvehicles/Open-Vehicle-Android #61: Graphic for Nissan e-NV200

glynhudson, 2018-08-01:
I'm successfully using the OVSM connected to my e-NV200 Nissan EV van. It uses the same EV electronics as a Leaf. All works great, however the OVSM app does not have a graphic for the e-NV200. I will try and contribute a graphic. Any tips on how the graphics are generated?

I will try and copy the graphics currently in use and create one for the e-NV200 https://github.com/openvehicles/Open-Vehicle-Android/tree/master/OpenVehicleApp/src/main/res/drawable-nodpi

[image]

dexterbg, 2018-08-01:
Go ahead. You need to provide:
- at least one car_<type>_<color>.png for the main/battery page
- at least one ol_car_<type>_<color>.png for the car info page
- optionally: overlays for the ol_car image to signal charging / open doors / … (see ol_car_twizy_chargeport.png for an example)
- at least one map_car_<type>_<color>.png icon for the map and notifications

Note: the images won't be used automatically, they need to be added to the modules.
Note 2: there's an open todo to separate these image sets from the app…

glynhudson, 2018-08-09:
Thanks for the guidence, very useful. I've made some progress: 

I've got a 'car' graphic of the env200 up and running 

https://github.com/glynhudson/Open-Vehicle-Android/commit/a720cfa3f3e91e3d4f00ba365f643dc68abe35fc

However when I select it the app crashes. This is paresumably since the top down view and the map view are missing. My plan was to use the leaf top-down graphic and map icon since the env200 and white leaf look the same from above! Where is the referance to tell the app to use the top-down view from the leaf?  

[image]

glynhudson, 2018-08-09:
Ok, I just duplicated and renamed the leaf graphics and it's working now  :+1: 

PR submitted https://github.com/openvehicles/Open-Vehicle-Android/pull/65

[image]

commit d0c6a60e (2018-08-09): Merge pull request #65 from glynhudson/add-env200

Add env200 Fixes #61
closed 2018-08-09 by dexterbg