openvehicles/Open-Vehicle-Android #62: Display AC status and cabin temperature on the Car page?

glynhudson, 2018-08-07:
Looking at the OVMS metrics list it looks like my Nissan Leaf is reporting its cabin temperature and climate control status:

`v.e.cabintemp`. 
`v.e.heating`   
`v.e.hvac`

These metrics are not accessible anywhere on the app. Has it been discussed to display the cabin temperature / AC status on the Car page of the app? It would be very useful to monitor the remote climate control status and the cabin temperature. Currently, after activating the remote climate control, there is no way to know if the remote climate is actually running or not.

dexterbg, 2018-08-08:
The A/C button indicates the A/C status, see capabilities discussion on why the button is missing.

The cabin temperature is a new metric and not yet included in the v2 protocol.

dexterbg, 2018-08-19:
I've added the cabin temperature to the "D" message:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commit/3e5224fc006dd8cab0661ad31e6320889ec887f2
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System/commit/66ffb5bdbfaab24326b437892c1397964d2e392b

You can now add it to the App.

dexterbg, 2018-08-20:
Remote updates are possible by http. Check out the `ota` command and the firmware web UI. You can download your own builds from any (e.g. local) web server. Or you can update from the official `edge` tag, that's the latest nightly build.

dexterbg, 2018-09-04:
I'll update the main & eap builds on my server this evening.

ltjessem, 2018-11-15:
> You can now add it to the App.

How does one do that? I'm using dexters-web and the lastest play store version of the app, but I'm not getting an A/C button or cabin temp. 
Is there a way to add metrics to the app that I'm missing?

glynhudson, 2018-12-22:
@dexterbg would you still consider merging a PR to add cabin temperature to the car page if I crated one? This has been my intention, sorry I'm been busy with other projects.

Im also keen to add a proper button for leaf A/C control rather than using home link 1-2. I'm not where to start with this. Is this something which is currently being worked or would you welcome a PR?

dexterbg, 2018-12-22:
Yes, of course. Adding scripts is not meant for the everyday user.

A/C control has been defined as MP command 26 some years ago, and that command has been introduced for the Leaf and is assigned to the A/C button in the App. Not sure what you intend to change there or why you would need to use homelink other than because of the missing capability message from a V3 module.

So the right way to fix this could be adding capabilities to V3 -- see mailing list on Mark's thoughts -- or to replace the caps check by a simple car type check as is done with other functions… ;)

ltjessem, 2018-12-23:
@glynhudson I have the A/C button working now after making the change to the V2 code described here: http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/005343.html  

A cabin temp indicator would be awesome!

dexterbg, 2018-12-23:
Please note:
> The following untested patch should make it work, but is obviously 
> inappropriate because the capabilities depend on the vehicle.

Adding a vehicle callback for the V2 capabilities is simple, but Mark had some V3 ideas on this:
http://lists.openvehicles.com/pipermail/ovmsdev/2018-July/005344.html

closed 2021-06-17 by glynhudson