openvehicles/Open-Vehicle-Android #68: Notifications don't work on newer versions of Android

glynhudson, 2018-08-12:
Testing the app on android 8 and android 9 the notifications don't work. They seem to work fine on android 7.1.1. 

Maybe this is a permissions issue? I'm aware that the way android permissions are handled changed on android 8.

dexterbg, 2018-08-14:
Maybe it's just the missing channel?
https://stackoverflow.com/questions/43093260/notification-not-showing-in-oreo

glynhudson, 2018-08-17:
Yes I think your right. However I think I've managed to add a notification channel (see: https://github.com/glynhudson/Open-Vehicle-Android/commit/0214dcd149a2fce0833fbf5526679dd21264e084). Is there anywhere to manully trigger a notification in ovms to test this? 

However notifications are still not working I think this is due to messages not showing up either in Android 8. . I'm still trying to understand how notifications work on OVMS. As far as I understand an incomming message triggers a notification, then the messages get stored in a database and displayed. Any ideas as to what could be stopping the messages coming through?

## Android 7

All messages automatically appear, and also trigger notifications:

[image]

## Android 8 

The only messages that appear are ones that are manually triggered, e.g responce to `stat` command:

[image]

dexterbg, 2018-08-19:
> Is there anywhere to manually trigger a notification in ovms to test this?

Yes: use the `notify.pl` client script (v2 repository). To get around the dupe filter in the App, change the text on successive calls.

> I'm still trying to understand how notifications work on OVMS. As far as I understand an incoming message triggers a notification, then the messages get stored in a database and displayed.

- The car (or any client in fact, see `notify.pl`) sends a push message (code `P`) to the server.
- The server forwards the message to all registered notification channels, which may include one or more GCM channels.
- The GCM server forwards the message to the App's `GcmListener`
- The `GcmListener` adds the message to the App database and creates the notification

For this to work, the App must have registered at the GCM server before, which is done by the `RegistrationIntentService` and the `InstanceIDListener`.

When sending commands from the message page, the results are received directly within the command result, so there is no GCM transport involved.

> Any ideas as to what could be stopping the messages coming through?

Check the App log to see if the GCM registration succeeds, if the GCM messages get to the App and if a notification is created. If the notifications don't appear in the message list after restarting the App, chances are there is something going wrong in the GCM registration.

glynhudson, 2018-08-22:
Thanks for explination. 

>Yes: use the notify.pl client script (v2 repository). To get around the dupe filter in the App, change the text on successive calls.

Thanks, I've manafed to populate `notify.pl` with my server and login detail and run it. The scrtipt outputs `usage: notify.pl text` but no notifications seem to be created. Is the `notify.pl` working correclty, is this expcted? 

>Check the App log to see if the GCM registration succeeds, if the GCM messages get to the App and if a notification is created. If the notifications don't appear in the message list after restarting the App, chances are there is something going wrong in the GCM registration.

Here is the log showing GCM refistration. It looks like it's managing to register ok. However the message list is not being shown

```
08-22 17:19:40.268 14756-14756/com.openvehicles.OVMS D/MainActivity: mApiObserver: login successful
    starting RegistrationIntentService for vehicleId=VN15WEF, gcmSenderId=497749188588
08-22 17:19:40.293 14756-14827/com.openvehicles.OVMS I/gcmRegIntentService: doing registration for vehicleId=VN15WEF, gcmSenderId=497749188588
08-22 17:19:40.302 14756-14827/com.openvehicles.OVMS I/gcmRegIntentService: got registration token: cU2C0oeEH2w:APA91bH_Ufw5eFMxxxxxxxxxxxxxxxxxxKCARMPCD1_Fe6Cu5obSSGZRQHqWM0hSH6uyzjf7bA
08-22 17:19:40.304 14756-14827/com.openvehicles.OVMS W/GCM: GCM SDK is deprecated, com.openvehicles.OVMS should update to use FCM
08-22 17:19:40.307 14756-14756/com.openvehicles.OVMS D/mGcmRegReceiver: onReceive intent: Intent { act=registrationComplete (has extras) }
    subscribing vehicleId=VN15WEF
08-22 17:19:40.309 14756-14756/com.openvehicles.OVMS D/GcmDoSubscribe: trying to subscribe vehicleId VN15WEF
    login OK, subscribing vehicleId VN15WEF on gcmToken cU2C0oeEH2w:APA9xxxxxxxxxxxxxxSH6uyzjf7bA
08-22 17:19:40.309 14756-14756/com.openvehicles.OVMS I/ApiTask: TX: MP-0 pae9706d9-d4ec-472c-88d7-985069f370f1,gcm,production,VN15WEF,EiPhu2ah,cU2C0oeEH2w:APA91bH_Ufw5eFMK4jrk45aERNOERcSyo0F1mxddQ7T9J0Uq20TgCKuYFmhZEVRV-WFGC7Kxxxxx2QOKaDeqdmWjQEQ2x92sZoFdsXkVfhId-KCARMPCD1_Fe6Cu5obSSGZRQHqWM0hSH6uyzjf7bA
08-22 17:19:40.310 14756-14756/com.openvehicles.OVMS D/GcmDoSubscribe: push subscription done.
```
It souds like GCM has been deprecated and replaced by FireBase cloud messenging (FCM). GCM will stop working work until April 2019: https://developers.google.com/cloud-messaging/faq. Implemeting the change to FCM will require changes to the server code as well as the app.

glynhudson, 2018-08-22:
Ah ok, sorry I didnt realise the notification text needed to be sent as an argument when calling the script. I know get the message:

```
./notify.pl test
  Client detects server digest is invalid - aborting
```
I'm using `ovms.dextersweb.de` and have entered the following into `ovms_client.conf`

```
[client]
vehicle_id=VA15WEF
server_password=xxxxx
module_password=xxxxx
server_ip=188.138.75.229
```
Is this the correct IP address for ovms.dextersweb.de? I did an IP address lookup to find it. Why require an IP address rather than a hostname?

Are notifications working correctly for you on your android device? What version of Android are you running? 

>Google… those guys certainly know how to annoy developers by deprecating perfectly working APIs and services every 2-3 years. I'm getting really tired of this…

I know, it's frustrating. Especially in this instance where the decision to move seems to be mostly commercial since they acquired Firebase. Although they do state in the migration guide that all the protocols, API calls and even tokesn should migrate over without hardly any changes: https://developers.google.com/cloud-messaging/android/android-migrate-fcm#update-server-endpoints

dexterbg, 2018-08-22:
You don't need to enter an IP address, the host name will do just as well. The error message means your id and/or password are wrong. Are you sure `server_password` is set to the vehicle account password -- not your user account password?

Notifications & `notify.pl` do work for my devices, which are Android 6 & 7.

glynhudson, 2018-08-22:
Doh! My vehicle_id was incorrect. I now get and OK when notify.pl. Still nothing received phone app but this is progress :+1: 

Should all messages sent via notify.pl appear in the Messages section of the app whatever the content?

dexterbg, 2018-08-22:
> Should all messages sent via notify.pl appear in the Messages section of the app whatever the content?

Not all. If a new message appears to be a dupe of the last it should not appear. If in doubt about the server transport, register for email notifications (see vehicle configuration on my server).

> `08-22 20:59:49.623 3826-3899/com.openvehicles.OVMS D/ApiTask: RX: MP-0 PA12345test (Z7JYkKswS3cIKbbFNK16/Q==)`

That is not a GCM notification, it's the MP message received by the TCP task. That will only work when the ApiTask is running, that's the reason we need GCM.

glynhudson, 2018-08-23:
I'm pretty sure the server transport is working since I can see the message being received in the debug log on the app. I've included the log output before and after for context. 

```
08-23 02:41:01.700 3643-3643/com.openvehicles.OVMS D/FragMap: after: addMarkers avail=758
08-23 02:41:01.729 3643-3643/com.openvehicles.OVMS D/FragMap: after: addMarkers added=758
08-23 02:41:01.730 3643-3643/com.openvehicles.OVMS I/FragMap: update: adding range circles: ideal=46.0 estimated=0.0
08-23 02:41:01.733 3643-3643/com.openvehicles.OVMS D/MainActivity: getdata: cache valid for lat/lng=53/-4, lat_main=53.142628
08-23 02:41:12.530 3643-3694/com.openvehicles.OVMS D/ApiTask: RX: MP-0 PAthis is a test notification (p8uWpQfmW9y2lW8KN8frt+xGXn0yjwnAXfEMrJCUDWzqXg==)
08-23 02:41:12.530 3643-3694/com.openvehicles.OVMS I/ApiTask: handleMessage: PAthis is a test notification
08-23 02:41:12.530 3643-3694/com.openvehicles.OVMS V/ApiTask: P MSG Received: Athis is a test notification
08-23 02:41:12.530 3643-3694/com.openvehicles.OVMS I/ApiTask: Push notification received: Athis is a test notification
08-23 02:41:12.531 3643-3643/com.openvehicles.OVMS D/ApiService: onUpdateStatus P
08-23 02:41:12.833 3643-3643/com.openvehicles.OVMS D/InfoFragment: Last updated: 11 secs ago
08-23 02:41:12.835 3643-3643/com.openvehicles.OVMS D/CarFragment: updateCarType: old=NL, new=NL
    Last updated: 11 secs ago
08-23 02:41:12.837 3643-3643/com.openvehicles.OVMS D/FragMap: update: Car on map: lat=53.142628 lng=-4.154234 maxrange=46.0Miles
    after: clearmap=true
08-23 02:41:12.842 3643-3643/com.openvehicles.OVMS D/FragMap: after: addMarkers avail=758
08-23 02:41:12.869 3643-3643/com.openvehicles.OVMS D/FragMap: after: addMarkers added=758
08-23 02:41:12.870 3643-3643/com.openvehicles.OVMS I/FragMap: update: adding range circles: ideal=46.0 estimated=0.0
08-23 02:41:12.871 3643-3643/com.openvehicles.OVMS D/MainActivity: getdata: cache valid for lat/lng=53/-4, lat_main=53.142628
```

If I force a message e.g send `stat` from within the app the log output is as follows:

```
08-23 02:42:49.343 2127-2657/com.android.systemui D/EGL_emulation: eglMakeCurrent: 0xafffcfa0: ver 2 0 (tinfo 0xaa31e530)
08-23 02:42:51.053 2346-2346/com.android.inputmethod.latin W/RichInputConnection: Slow InputConnection: GET_WORD_RANGE_AT_CURSOR took 406 ms.
08-23 02:42:52.325 1778-2679/system_process I/AccountManagerService: getTypesVisibleToCaller: isPermitted? true
08-23 02:42:55.928 3643-3643/com.openvehicles.OVMS D/NotificationsFragment: initUi: (re-)loading notifications
08-23 02:42:55.928 3643-3643/com.openvehicles.OVMS D/OVMSNotifications: Loading saved notifications list from database
08-23 02:42:55.929 3643-3643/com.openvehicles.OVMS D/OVMSNotifications: Loaded 7 saved notifications
08-23 02:42:55.930 3643-3643/com.openvehicles.OVMS I/ApiTask: TX: MP-0 C7,stat
08-23 02:42:55.997 3643-3689/com.openvehicles.OVMS D/EGL_emulation: eglMakeCurrent: 0xaaa83cc0: ver 2 0 (tinfo 0xaaabfd70)
08-23 02:42:56.193 3643-3689/com.openvehicles.OVMS D/EGL_emulation: eglMakeCurrent: 0xaaa83cc0: ver 2 0 (tinfo 0xaaabfd70)
08-23 02:42:56.236 3643-3694/com.openvehicles.OVMS D/ApiTask: RX: MP-0 c7,0,Not chargingSOC: 53.0%Ideal range: 46MODO: 34595.0MCAC: 57.4AhSOH: 87% (IUs4Fahu3qmQ32kkDckQbN/CcaWa90Q4L5Hg/fSLs6c7u48UEJIA0xngc/l0cVvRW1hHZ5mMtSDEWa3dofo5Y4R4pFt4Wb0NgnScS/BzFwYCnKgPBeo=)
08-23 02:42:56.236 3643-3694/com.openvehicles.OVMS I/ApiTask: handleMessage: c7,0,Not chargingSOC: 53.0%Ideal range: 46MODO: 34595.0MCAC: 57.4AhSOH: 87%
08-23 02:42:56.236 3643-3694/com.openvehicles.OVMS V/ApiTask: c MSG Received: 7,0,Not chargingSOC: 53.0%Ideal range: 46MODO: 34595.0MCAC: 57.4AhSOH: 87%
08-23 02:42:56.236 3643-3694/com.openvehicles.OVMS I/ApiTask: Command response received: 7,0,Not chargingSOC: 53.0%Ideal range: 46MODO: 34595.0MCAC: 57.4AhSOH: 87%
08-23 02:42:56.249 3643-3643/com.openvehicles.OVMS D/NotificationsFragment: initUi: (re-)loading notifications
08-23 02:42:56.249 3643-3643/com.openvehicles.OVMS D/OVMSNotifications: Loading saved notifications list from database
08-23 02:42:56.255 3643-3643/com.openvehicles.OVMS D/OVMSNotifications: Loaded 8 saved notifications
```

It seems the difference is here is the log energy ` Loading saved notifications list from database` this does not happen when a push notification is received. Could you post the log output of a succesfully processed push notification on one of your devices? 

I'm running the standard Master branch on Androud 6.0. It seems this issue is not only specific to Android 8 and above. 

Only messages that I manually trigger appear in the message list and notifications don't work:

[image]

dexterbg, 2018-08-23:
GCM definitely still works, I've got no reports notifications stopped getting through.

In the log you should see info level entries from tag `MyGcmListenerService` like `Notification received from…`.

Have you double checked the GCM sender ID?

glynhudson, 2018-10-03:
Are notifications still working for you? I've been unable to get notifications to work now with any of my devices running the standard google play version. 

>In the log you should see info level entries from tag MyGcmListenerService like Notification received from….

In my log file (see above) I'm not getting any entries from `MyGcmListenerService`. 

>Have you double checked the GCM sender ID?

My GCM sender ID field in the app was blank, I assumed (default = blank) means that this field should be left blank. App seemed to work fine without it. I've just populated the GCM sender ID with my ID from the web server, I'm still not totally sure that this is or why it's needed. Should it be required for notifications? Either way notifications are still not working for me :-( 

I'm not sure what else I can try and test? Have any other users reported this issue?

dexterbg, 2018-10-03:
The GCM ID needs to be the one of the OVMS server you're connected to. The default ID (blank) only applies to the Server api.openvehicles.com. If you're using my server (ovms.dexters-web.de), it needs to be 1043773844049. If you've setup your own server, it would be the ID you got from your GCM sender registration. After changing the ID you may need to terminate and restart the App.

dexterbg, 2018-10-03:
And no, I still had no other reports regarding this issue.

glynhudson, 2018-10-04:
Whoo! Setting the GCM ID has made the notifications work :-)

I feel very silly for wasting your and my time trying to debug this. Although in fairness there was no mandate to set the GCM ID and the app seemed to work fine without it so I was none the wiser. I just assumed it should be kept blank as default.

I think a note on the setting page of the app might have helped me realise this sooner. I'm happy to try and work a PR for this if we could decide on the wording? 

e.g "enter GCM ID from you OVMS server, required for push notifications>" 

Would it be possible for the app to obtain the GCM ID automatically from your server since the ID is the same for all users on your server?

closed 2018-10-04 by glynhudson