openvehicles/Open-Vehicle-Android #74: e-Niro charge amps on new row.

Timopen, 2019-05-07:
When charging over 100 A the amps are moved to a new row and no longer visible.
[image]

glynhudson, 2019-05-29:
This should be an easy fix, I'll try and take a look. Would you mind posting a non annotated screenshot of the Kia eNiro rapid charging? Would you mind if I used this screenshot for a flyer for OVMS that I'm making? 

I'm currently using a leaf (my car) as the main car, however I would like to change this to be a eNiro but it would look silly to show a eNiro graphic with the estimated range of my leaf! A real eNiro screenshot would be amazing. Hope you're enjoying it, it looks like a fantastic car from everything I've read. 

[image]

Timopen, 2019-05-30:
Hi Glyn
I don't have a picture without annotation, but I promise I will take one next time I fast charge. I can use the Ionity charger nearby and we might have Amps near 200 in the picture. Meanwhile here is a slow charge picture.
[image]
And overview:
[image]

glynhudson, 2019-06-08:
Thanks, that's great. Does climate control pre heating / cooling work on eNiro for you?

glynhudson, 2019-06-08:
Climate control my require using home link 1 to activate

Timopen, 2019-06-09:
No preheating/cooling on the KIA's yet.

glynhudson, 2019-06-09:
Thanks, that's good to know. Is it not even on the Soul? I thought pre heating was working on the Soul?

Timopen, 2019-06-10:
No, not on Soul either. The info on shop.openenergymonitor.com is a bit missleading at the moment.

glynhudson, 2019-06-10:
Thanks, apologies for the confusion. It's now been fixed. Thanks for your help.

https://shop.openenergymonitor.com/open-vehicle-monitoring-ovms-wifi-3g-europe/

glynhudson, 2019-06-10:
What about TPMS, ODO, speed display (km/h) and door unlocking? Are these features supported on the eNiro? I believe these features are supported on the Soul

Timopen, 2019-06-11:
All the above exept speed is working. I'm going to try to fix the speed eventually, it shows a bit strange values at the moment. How is the fix for the amps on the same row in the app going?

glynhudson, 2019-06-12:
@Timopen thanks, that's very useful.

I've attempted to fix the text wrapping by enlarging the text box. However, I have been unable to recreate the text wrapping issue on my device (pixel 2). What device are you using? What is the screen width? 

https://github.com/openvehicles/Open-Vehicle-Android/pull/76

Timopen, 2019-06-15:
Glyn
I'm using a Oneplus 6T 2340 x 1080 pixels 402 ppi Android 9 Pie.
When usin my old phone Sony xperia 1080 x 1920 pixels, 16:9 ratio (~424 ppi density) Android 8 Oreo there is no wrap.
Oneplus:
[image]

Timopen, 2019-06-15:
Sony:
[image]

Timopen, 2019-06-21:
Hi Glyn
I noticed on the open energy monitor site, there is a selection for KIA OBD2 cable.
This should read KIA Soul OBD2 cable as the e-Niro uses the generic one.

Med Vänlig Hälsning

TIMO PENTTILÄ

#76<https://github.com/openvehicles/Open-Vehicle-Android/pull/76>

closed 2019-06-23 by dexterbg

glynhudson, 2019-06-23:
@timopen thanks for letting me know. I will correct the listing and update docs.

glynhudson, 2019-07-04:
>This should read KIA Soul OBD2 cable as the e-Niro uses the generic one.

@Timopen do you know if the Kia Niro required 'left' or 'right' generic OBD2 cable? Or would either fit

commit 15adb6b8 (2019-06-23): attempt to fix charging text display width fix #74
commit 4150a26d (2019-06-23): fit charging current fix #74
commit a97efd5e (2019-06-23): Revert "fit charging current fix #74"

This reverts commit 4150a26d068ab73fee6ef2aeb9eb80b010c04e81.