openvehicles/Open-Vehicle-Android #75: e-Niro graphics mising?

glynhudson, 2019-05-29:
Nice work @dexterbg adding app support for the eNito, this is a very exciting car to support. Lack of app support will make OVMS a very useful addition for many owners.  

Just looking at this commit https://github.com/openvehicles/Open-Vehicle-Android/commit/c155615a5def1a7baaf54804b9c978a34323dc07, I can't see any graphic resources included for the eNiro even though `map_kianiro_xxx` and `car_kianiro_gray` are referenced in the code. Did you by any chance forget to add the image files to the git commit?

dexterbg, 2019-05-30:
Very strange, the Android Studio VCS tool seems to have missed them. The APK build did include them though…

closed 2019-05-30 by dexterbg

glynhudson, 2019-05-30:
Thanks, Fixed by https://github.com/openvehicles/Open-Vehicle-Android/commit/85bd52c0c07211069933678f54ef69af2197c424