openvehicles/Open-Vehicle-Android #79: Latest version app crashes

Timopen, 2019-09-08:
Version 3.12.11 crashes on opening on my Oneplus 6T running Android 9.
[image]

Timopen, 2019-09-10:
Here is the log from Android studio debug:
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.openvehicles.OVMS, PID: 25383
    java.lang.IllegalArgumentException: radius is negative
        at com.google.maps.api.android.lib6.common.l.d(:com.google.android.gms.dynamite_mapsdynamite@19056081@19.0.56 (100400-262933554):12)
        at com.google.maps.api.android.lib6.impl.am.<init>(:com.google.android.gms.dynamite_mapsdynamite@19056081@19.0.56 (100400-262933554):12)
        at com.google.maps.api.android.lib6.impl.be.a(:com.google.android.gms.dynamite_mapsdynamite@19056081@19.0.56 (100400-262933554):334)
        at com.google.android.gms.maps.internal.k.a(:com.google.android.gms.dynamite_mapsdynamite@19056081@19.0.56 (100400-262933554):268)
        at cv.onTransact(:com.google.android.gms.dynamite_mapsdynamite@19056081@19.0.56 (100400-262933554):4)
        at android.os.Binder.transact(Binder.java:688)
        at com.google.android.gms.internal.maps.zza.transactAndReadException(Unknown Source:7)
        at com.google.android.gms.maps.internal.zzg.addCircle(Unknown Source:9)
        at com.google.android.gms.maps.GoogleMap.addCircle(Unknown Source:4)
        at com.androidmapsextensions.impl.GoogleMapWrapper.addCircle(GoogleMapWrapper.java:75)
        at com.androidmapsextensions.impl.CircleManager.createCircle(CircleManager.java:45)
        at com.androidmapsextensions.impl.CircleManager.addCircle(CircleManager.java:39)
        at com.androidmapsextensions.impl.DelegatingGoogleMap.addCircle(DelegatingGoogleMap.java:73)
        at com.openvehicles.OVMS.ui.FragMap.addCircles(FragMap.java:493)
        at com.openvehicles.OVMS.ui.FragMap.update(FragMap.java:462)
        at com.openvehicles.OVMS.ui.FragMap.update(FragMap.java:404)
        at com.openvehicles.OVMS.ui.BaseFragment.onServiceAvailable(BaseFragment.java:121)
        at com.openvehicles.OVMS.api.ApiObservable.notifyOnBind(ApiObservable.java:61)
        at com.openvehicles.OVMS.ui.ApiActivity$1.onServiceConnected(ApiActivity.java:48)
        at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1770)
        at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1802)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6898)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:537)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

Timopen, 2019-09-10:
Found a fix ! :)
Preventing rd1 and rd2 from being negative in FragMap fixed the issue.

// draw circle in a map
	private void addCircles(float rd1, float rd2) {
		if (map == null)
			return;
		if (rd1 < 0) rd1 = 0;
		if (rd2 < 0) rd2 = 0;

		float strokeWidth = getResources().getDimension(
				R.dimen.circle_stroke_width);
		List<PatternItem> pattern = Arrays.<PatternItem>asList(new Dot());

		// full range circles:

		map.addCircle(new CircleOptions()
				.data("first circle")
				.center(new LatLng(lat, lng))
				.radius(rd1 * 1000)
				.strokeWidth(strokeWidth)
				.strokeColor(Color.BLUE));

		map.addCircle(new CircleOptions()
				.data("second circle")
				.center(new LatLng(lat, lng))
				.radius(rd2 * 1000)
				.strokeWidth(strokeWidth)
				.strokeColor(Color.RED));

		// half range ("point of no return") circles:

		map.addCircle(new CircleOptions()
				.data("first ponr")
				.center(new LatLng(lat, lng))
				.radius(rd1 * 1000 / 2)
				.strokeWidth(strokeWidth / 2)
				.strokePattern(pattern)
				.strokeColor(Color.parseColor("#A04455FF")));

		map.addCircle(new CircleOptions()
				.data("second ponr")
				.center(new LatLng(lat, lng))
				.radius(rd2 * 1000 / 2)
				.strokeWidth(strokeWidth / 2)
				.strokePattern(pattern)
				.strokeColor(Color.parseColor("#A0FF5544")));
	}

dexterbg, 2019-09-10:
Nice, do you want to create a pull request or shall I add the fix in your name?

Timopen, 2019-09-10:
Please add the fix.

Timopen, 2019-09-11:
Haven't found the reason why the estimated range went crazy, but a soft reset fixed it. But it's still good to have the fix to prevent the app from crashing.

closed 2019-09-21 by dexterbg