openvehicles/Open-Vehicle-Android #82: AutoStart background service fails on Android 8

dexterbg, 2019-09-18:
Background TCP connectivity got even worse with Android 8.
From our crash reports:

```
java.lang.RuntimeException: 
  at android.app.ActivityThread.handleReceiver (ActivityThread.java:3514)
  at android.app.ActivityThread.access$1200 (ActivityThread.java:201)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1736)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:193)
  at android.app.ActivityThread.main (ActivityThread.java:6971)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:493)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:865)
Caused by: java.lang.IllegalStateException: 
  at android.app.ContextImpl.startServiceCommon (ContextImpl.java:1577)
  at android.app.ContextImpl.startService (ContextImpl.java:1532)
  at android.content.ContextWrapper.startService (ContextWrapper.java:664)
  at android.content.ContextWrapper.startService (ContextWrapper.java:664)
*** at com.openvehicles.OVMS.receiver.AutoStart.onReceive (AutoStart.java:25) ***
  at android.app.ActivityThread.handleReceiver (ActivityThread.java:3505)
```

Web infos:
- https://stackoverflow.com/questions/46445265/android-8-0-java-lang-illegalstateexception-not-allowed-to-start-service-inten
- https://developer.android.com/about/versions/oreo/background.html

mbernier85, 2019-09-19:
I guess from my pull request, you were talking about this issue here ?

Solution is to create a foreground service, this is pretty easy to fix.
I will need to dig a bit into the code to understand what is the purpose of the service in more details.

mbernier85, 2019-09-19:
From my first look, the issue is that you are starting a service from a Broadcast receiver, this is not possible anymore, you have to start from an Activity

dexterbg, 2019-09-19:
The service is used to communicate with the server. It needs to run as a service to provide car data updates and the command API to scripts, e.g. (mostly) Automagic widgets, independent of the App.

So the service must also start independant of the App when the device reboots, or custom scripts will not work.

dexterbg, 2019-09-19:
Yes, the service shall keep the TCP connection open 24x7.

A periodical job would need to do a full reconnect to the server on every run, so it wastes ressources both on the Android device and the server. It also won't allow users running 24x7 automation scripts, e.g. to dynamically control solar charge power.

Keeping the TCP connection up never was a battery drain or network volume issue with previous Android versions. Users were in control as Apps needed a power manager whitelisting to actually run 24x7.

So, no continuous background TCP connectivity in Android >= 7, not even with some Google cloud service? Android has lost this ability completely?

mbernier85, 2019-09-19:
nothing prevents you from running a background service, then use a work manager to recreate the service when it dies.

I'm sure there is a solution that could work most of the time.

But yeah Google pretty much killed background service, best way is a foreground service, but you won't be able to start on BOOT like it's currently done. But with a work manager it might be possible to make it work

mbernier85, 2019-09-20:
https://github.com/mbernier85/Open-Vehicle-Android/tree/fix-leak

you might want to test this branch, see if it fixes your service issue

But the biggest thing here is, I removed all the static ref to context.

dexterbg, 2019-12-14:
Solved in release 3.14.0

closed 2019-12-14 by dexterbg