openvehicles/Open-Vehicle-Android #90: OVMS Android Version 3.13.2 Crashing

JohnGarziglia, 2019-12-09:
The newest Android Version 3.13.2 (2019112701) crashes on my Pixel 3XL. Up to now, previous versions have run perfectly. When it first automatically installed itself, it wiped out my settings for my car leaving only the demonstration vehicle. Now, each time I try to add the settings for my car, it crashes and any subsequent attempt to open the app gives me an error box stating "Open Vehicles keeps stopping" and I am unable to subsequently open the app until I uninstall and re-install it.  I appreciate any advice .  I have sent in reports with the logs.

John Garziglia

dexterbg, 2019-12-09:
Please try this beta build: https://dexters-web.de/f/tw-beta/OpenVehicleApp-release-3.13.2-2019120901.apk

If that also doesn't work, please try clearing the App cache & data.

JohnGarziglia, 2019-12-09:
Hi Michael,

Clearing the App cache and data seems to have done it.  It is now working.
Thank you for your assistance.

All the best,
    John
___________________________________________

John Garziglia
11140 Glade Drive
Reston, VA 20191
(703) 476-0808
JohnGarziglia@gmail.com

dexterbg, 2019-12-14:
Solved in release 3.14.0

closed 2019-12-14 by dexterbg