openvehicles/Open-Vehicle-Android #99: App asks for passord for half a second upon opening it

dalathegreat, 2019-12-19:
Steps to reproduce

1. Open the app
2. Lock the phone
3. Wait for xx minutes
4. Unlock phone

When the app launches, it tells you that you need to login. This window appears for roughly half a second, and then disappears. Maybe some delay could be added so the app doesn't ask for password right away? This would improve the look and feel of the app quite a lot.

dexterbg, 2019-12-20:
Not sure what you mean. There is no message in the App telling you that you need to login, and the App doesn't ask for a password. The only dialogs asking for (kind off) passwords are the ones triggered by valet mode for example, they ask for a PIN. But they're not about any login.

Is it possible you mean the info dialog about the lost server connection, possibly resulting from the phone shutting down it's internet connection when entering sleep mode? That's automatically popped down as soon as the service has reestablished the connection. As long as the connection is down, most of the App won't be functional, so omitting the dialog will lead to a bad user experience.

You've got the option to keep the connection open. This will also work in sleep mode, if you configure the Android power savings App exception. I use this option all the time because I'm using some Automagic OVMS widgets on my home screen independent of the App.

dalathegreat, 2019-12-20:
That's it! It happens so fast that you sometimes don't have enough time to read it all. I'll try enabling that option!

closed 2019-12-20 by dalathegreat