openvehicles/Open-Vehicle-Android #123: No SSL support

basak, 2021-03-21:
Hi,

Thank you for OVMS!

I just bought one and am setting up my own server. I realised that the Android app doesn't seem to support SSL and confirmed it here by looking at the code:

https://github.com/openvehicles/Open-Vehicle-Android/blob/4170b6611ae0ec433345e072051f951fb1ac9482/OpenVehicleApp/src/main/java/com/openvehicles/OVMS/api/ApiTask.java#L276

dexterbg, 2021-03-22:
Correct. When adding SSL support, take care to make it optional per car, as some users won't have encryption enabled on their servers.

markwj, 2024-03-25:
> Should this issue report not be handled as a security issue instead of a advise how the code should be changed by the user?
> TLS 1.3 or equivalent encryption shoud be default everywhere this days.

Note that the OVMS protocol v2 has its own encryption layer (one mandatory and a second optional). The reason for this was that historically the microcontrollers used in earlier version of OVMS didn't have enough power to run TLS/SSL.

TLS/SSL support was fairly recently added on top of that very basic encryption, and has been implemented both in the servers and the latest vehicle modules.

This issue here is to add SSL support to the Android App, to supplement the basic v2 protocol encryption.

fowi4hjte, 2024-03-25:
Thanks for the reply.
Is there some planned time scale when this basic v2 protocol would be implemented? Maybe implement in the between a warning in the app, that all the data is been send out of the app without encryption. The user should know such important things to adapt their behavior on facts like this. 

After i spend some time reading deeper into the project here some additional informations:
@markwj wrote, that earlier versions of the OVMS did not have enough power to do encryption.
OVMSv1 was based on PIC 18F2580 and OVMSv2 was based on PIC 18F2685 based on the information here: https://www.openvehicles.com/aboutus
Since OVMSv3 switched to ESP32, things changed and make encryption possible. Even older ESP8266 was capable of proper encryption. Such chips can do WPA3 on their WiFI, TLS1.3 on their traffic and many other important encryption things.

Additional warning for the users: Those chips like every other computer in the world is just as safe as the amount of security fixes applied as fast as possible to close known security issues. At the moment this project is using outdated and now unsupported IDF. You can compare it simplified as an outdated operating system of a computer like running Windows 7 in 2024 on the internet. The ESP-IDF used by OVMSv3 is version 3.3. The current version is 5.2 https://github.com/espressif/esp-idf
There are already many known security issues that of course stay unfixed in the old version 3.3. Some examples: https://cve.mitre.org/cgi-bin/cvekey.cgi?keyword=esp32

markwj, 2024-03-25:
I think you misunderstand.

Protocol v2 has encryption. Just quite basic due to the capabilities of the microcontrollers it ran under when released.

If you have any remote exploits which are effective from the public Internet side, without physical access to the device, then please provide details including a proof of concept. For example, the list you provided starts with a bunch of CVEs on esphome and is irrelevant to the OVMS project.

Alternatively, if you or others are willing to step forward and help with the implementation of SSL/TLS on the Android app, on top of the existing protocol v2 encryption, as always that help would be appreciated.

fowi4hjte, 2024-03-25:
OVMS protocol v2 is already implemented (and the default/only way to use the app) in the current Android app version?
I probably misunderstood your first message and understand it the way, that OVMS protocol v2 have (its own) encryption, but the Android App currently is not using OVMS protocol v2 and have to be updated to use it.

markwj, 2024-03-25:
> OVMS protocol v2 is already implemented (and the default/only way to use the app) in the current Android app version?
Yes

> I probably misunderstood your first message and understand it the way, that OVMS protocol v2 have (its own) encryption, but the Android App currently is not using OVMS protocol v2 and have to be updated to use it.
No

The current Android app uses protocol v2 with its own basic encryption. This GitHub issue is to also add ssl/tls on top of that.