openvehicles/Open-Vehicle-Android #129: Release on F-Droid

aha999, 2021-06-25:
If your app was released on F-droid, where most of open source android apps are, it could get more users involved.

Please release the app on F-droid.

vazhnov, 2024-01-26:
First, the license should be clear: https://github.com/openvehicles/Open-Vehicle-Android/issues/157.

dexterbg, 2024-01-27:
From a quick glance, F-Droid requires quite some additional setup & maintenance effort and a Gitlab account with annual costs, so that's not likely to happen. Releasing on Google Play is free for us, and all we need to do is build & upload the package.

closed 2024-01-27 by dexterbg

ramack, 2024-03-02:
I didn't see this ticket before creating #163, sorry about that. Finally it's up to you to decide if you want to publish on f-droid, but I wanted to remark, that you can integrate the f-droid release fully automated simply by tagging a release here on github, without any additional cost. Doing the initial setup is clearly some effort, but then it should be easy going - even less effort than for a release on GPlay.

After a quick look into the dependencies I guess the biggest effort for a f-droid release would be to free the app from closed-source libraries and services. Anyway, I don't plan to use the OVMS Android App, so even I think that would be the best approach, it's fine for me.

fowi4hjte, 2024-03-24:
Could this get reopened? F-Droid release does not require a gitlab account and also does not require annual costs.

markwj, 2024-03-25:
If a developer wants to step forward and take on the project, we can re-open this issue. At the moment, it seems there are two people interested in the app on f-droid, but which will actually do the work?

relan, 2024-03-25:
F-Droid maintainer here.

1. Publishing an app on F-Droid does *not* require a GitLab account. There are maintainers and casual contributors who maintain apps metadata on the catalogue.

2. F-Droid does *not* collect any fees. GitLab does not require a paid account either.

3. The only *required* thing from an app author is a permission to publish their app on F-Droid. An F-Droid maintainer or another contributor can do that.

4. The main obsticle for OVA inclusion into F-Droid are proprietary libraries:

```
    implementation 'com.google.android.gms:play-services-maps:18.2.0'
    implementation 'com.google.android.gms:play-services-gcm:17.0.0'
    implementation 'com.google.firebase:firebase-messaging:23.0.2'
```

F-Droid requires all apps to be 100% free software (this is the main repo policy, there are other F-Droid repos with different policies).

licaon-kter, 2024-03-25:
F-Droid recently started adding apps from Forum discussions: https://forum.f-droid.org/tag/leavegitlab _(the tag is an inside joke related to https://gitlab.com/fdroid/admin/-/issues/159)_

It works better with the Gitlab CI, since you can see the app building and failing, but if Gitlabs behaviour has gotten worse the Forum will pickup the apps as needed.

markwj, 2024-03-25:
As discussed, no objections here to somebody stepping forward to do the work to port to f-droid - but I would caveat that with not affecting the experience of other users not using f-droid.

It just seems not a priority / interest for the main app maintainers here.