openvehicles/Open-Vehicle-Android #132: Android intent?

glynhudson, 2021-10-12:
Hi, I'm trying to follow the [docs on the wiki](https://github.com/openvehicles/Open-Vehicle-Android/wiki) to use an Android Intent Action to get the ovms android app to send a message to a vehicle to start a custom script. 

I'm using Tasker to send the intent but I'm not having any luck so far, any ideas what I'm doing wrong? I'm also sending a intent with the service password and vehicle ID 

[image]

dexterbg, 2021-10-13:
Glyn, this is really not a Tasker support forum… all I can do is show you the respective Automagic action definition:

```
  <action type="send_broadcast">
    <useDefaultName>false</useDefaultName>
    <name>SetProfil</name>
    <intentAction>com.openvehicles.OVMS.SendCommand</intentAction>
    <categoryListEnabled>false</categoryListEnabled>
    <categoryList>android.intent.category.DEFAULT</categoryList>
    <dataEnabled>false</dataEnabled>
    <data></data>
    <dataTypeEnabled>false</dataTypeEnabled>
    <dataType>text/plain</dataType>
    <explicitComponent>false</explicitComponent>
    <packageName></packageName>
    <className></className>
    <flagList>FLAG_ACTIVITY_NEW_TASK</flagList>
    <extras>putString("sel_vehicleid", "XXXX");
putString("sel_server_password", "XXXX");
putString("msg_command", "7,script eval abrp.send(1)");
</extras>
    <waitForResult>false</waitForResult>
    <accessResultExtras></accessResultExtras>
  </action>
```

From your log it appears Tasker only adds the "msg_command" extra.

glynhudson, 2021-11-02:
Thanks, this worked great 👍 

[image]

closed 2021-11-02 by glynhudson