openvehicles/Open-Vehicle-Android #139: possibility of having a more precise soh?

cyrelec86, 2022-05-19:
Hello the community I wanted to know if it was possible to have a soh with 2 decimal places? I am doing a study on the degradation of the 40 kw battery of the Léaf and I need a precise soh! do you think it is possible?
thank you in advance for your answer 😉

glynhudson, 2023-03-29:
I don't think the SOH reading from the BMS is accurate enough to merit two decimal places! The SOH is a rough estimation of a chemical process, it's a very rough estimation at best. I wouldn't even consider looking at even one decimal place. The average SOH over a rolling 12-month period is a better measurement. Here's a 3yr graph os SOH of my 2014 Leaf, now with 90,000 miles. 

[image]

dexterbg, 2023-03-30:
Correct. Adding decimal places suggests precision and accuracy where there is really none.

If you're interested in the raw value the vehicle provides, you can always do `metrics list soh`. Some vehicle adapters provide multiple indicators, check their documentation for custom metrics, but the "soh" filter will probably already include them.