openvehicles/Open-Vehicle-Android #140: Individual battery cell health

kallisti5, 2022-06-18:
It would be nice (tm) if OVMS also supported showing individual cell voltages in addition to the overall pack SoC.  The cell voltages should be available via OVMS. A delta could be calculated as well showing differences between the min and max cell voltage.

kallisti5, 2022-06-18:
Here's an example from the popular LeafSpy for the Nissan Leaf..

[image]

dexterbg, 2022-06-19:
You're welcome to take care of this.

The App already has a chart module for this, I implemented this for the Twizy some years ago when still on V2 hardware. The Twizy BMS subsystem was the origin for the generalized V3 BMS subsystem.

The Twizy has only 14 cell voltages & 7 temperatures. I added two custom history records for this, "RT-BAT-P" & "RT-BAT-C", with each cell corresponding to one record key and updates only being sent on changes, so this could fit into the V2 module.

The Twizy chart data receiver thus needs to reassemble the diff records into a full history, and the chart then allows to browse the 24 hour cell history using the knob at the bottom:
https://dexters-web.de/faq/en#whatdoes
https://dexters-web.de/g/custom/ovms/fn/Battery-Monitoring.png

The App chart can adapt to more cells, it allows zooming and scrolling similar to the web UI. So you can use that as the starting point, it's this App module:
https://github.com/openvehicles/Open-Vehicle-Android/blob/master/OpenVehicleApp/src/main/java/com/openvehicles/OVMS/ui/BatteryFragment.java

The "RT-BAT-C" record structure is not suitable for battery packs consisting of many cells. A general record structure could adopt the current metrics array structure (the ``v.b.c.`` metrics), but then needs to drop the diff based data reduction. That's ok if the user can configure a suitable update interval to keep the data volume low.

Another option is to drop the history feature for the first implementation and only display the current status, as does the web UI battery chart. The web UI battery chart reads the BMS metrics updated via the websocket channel, so implements a live view, similar to LeafSpy, but more powerful as it integrates most of the available cell details. Btw, this works on a mobile phone display as well, as the chart also adapts to the available size and has pinch zooming.

The App could query the metrics via command execution (``metrics list v.b.``), so no dedicated record structure is needed for this. The downside to this approach is, it will need a live module connection.

Dropping the history browsing and using the current status only would mean the App's chart has no advantage over the web UI though, yet also lacks the actual live view capability. So I think it would make sense to base this on generalized BMS MP data records from the beginning.

Regards,
Michael