openvehicles/Open-Vehicle-Android #141: UI refresh

kallisti5, 2022-06-18:
The UI could use some modernization. I'd like to see more technical information over so much space used for a picture of a charging plug.

Here's my Bolt EV for an example:

[image]

I think the space could be better utilized

dexterbg, 2022-06-19:
While the UI certainly can use some modernization, you're relating to the most basic view configuration here. The InfoFragment adapts to the current status and available data and can already become a bit crowded during a charge with a vehicle providing all metrics.

Also, adding more info to a UI generally isn't necessarily good for normal users. IMO the main UI should be kept simple, focusing on the info & controls of most common interest. More detail can be provided on additional pages / optional overlays.

For example, the battery SOH & capacity info is a detail a normal user isn't interested in, except when buying or selling a car. Therefore it's located on the additional vehicle info page reachable via the dropdown menu.

Regards,
Michael

kallisti5, 2022-11-27:
@dexterbg any thoughts on putting the overall battery SoC on the car page?      I could see the Car page having an overview of everything, and the Battery page giving a more detailed view of Battery stuff (SoC, Temp,  etc)

Using this example, adding the SoC would enable a "one page summary" of everything about your car.
[image]

kallisti5, 2022-11-27:
Example mock up:
[image]

developerfromjokela, 2024-12-24:
[image] 
[image]

Hi and merry christmas to everybody. I'm working on this kind of layout. under the car visualisation will be quick functions and after that I plan to put list of sections.  Pardon me for the bad rim image, I took snapshot while wheel was animating movement :D I've also made support for animating a moving vehicle, showing body parts and states, etc. All based on drawables so not 3d rendering.

developerfromjokela, 2024-12-24:
#43  check out sketch :)

developerfromjokela, 2024-12-24:
Obviously the graphics are still sketch and raw, but the concept is working. I dove into rendering cars as 3d but 1) rendering sucks on android 2) most of car models are available as paid options, 40+€ each :D