openvehicles/Open-Vehicle-Android #144: ms_v_charge_type->SetValue("type1") shows J1772 type2 in Android app

kallisti5, 2022-11-26:
I just noticed that setting "type1" in OVMS for ms_v_charge_type still shows a type2 connector in the app.

It would be interesting to add some logic around "type1  level 1 vs type2 level 2" in the app and render each plug correctly.

kallisti5, 2022-11-26:
Example:

[image]

kallisti5, 2022-11-26:
Actually... I just realized this is because I selected the Zoe vehicle.   These images are static.   Would it be beneficial to "standardize" the charge connector icons and render them in the needed locations instead of just templating out the whole vehicle?

dexterbg, 2022-11-27:
Beneficial in terms of App size if we would replace the existing overlay images, but not beneficial in terms of developer & user contentment then: overlays for some vehicles have been carefully designed to fit the vehicle charge port and facilities.

We could provide a set of default icons for cars without dedicated overlays though.

kallisti5, 2022-11-27:
Yeah, you're right on this one.   I added some specific overlays for the Bolt EV and it was easy enough.

closed 2022-11-27 by kallisti5