openvehicles/Open-Vehicle-Android #145: Repeated "OpenChargeMap read failed" messages

basak, 2022-12-28:
Hi,

After a recent app update I'm getting repeated "OpenChargeMap read failed" messages. I don't really want this functionality anyway, as it leaks my location. But I can't find any option to turn it off. I don't know why it's failing, but either way I guess I shouldn't be getting repeated messages, and I hope it's not spamming the service?

dexterbg, 2022-12-28:
You're right, there should be an option to disable this feature.

The Google Maps integration also generally leaks your location. Would you rather want to have an option to completely disable the map?

Regarding the "read failed" messages, I can reproduce these over here by moving the map to an uncached area. Probably a temporary issue at the OCM server, may also be a temporary blocking of our API key due to many clients now fetching updates.

basak, 2022-12-28:
> You're right, there should be an option to disable this feature.

I'd appreciate that - thanks!

> The Google Maps integration also generally leaks your location. Would you rather want to have an option to completely disable the map?

I find the location map really useful, so even though I don't like leaking location information to Google, I'd probably keep it. Pragmatically, they have my location data anyway as I'm using an Android phone with Play Services enabled. Ideally though, I'd like integration with OsmAnd to keep everything local. Something similar to https://www.osmand.net/docs/user/plugins/osmand-tracker/.

> Regarding the "read failed" messages, I can reproduce these over here by moving the map to an uncached area. Probably a temporary issue at the OCM server, may also be a temporary blocking of our API key due to many clients now fetching updates.

Ah - I thought it might be a permanent problem. Maybe you could rate-limit the failure notices though, with a final message of "further messages suppressed" or something like that?

If it's transient though and not really regressed by the update, then there's no urgency. Thank you for your continued work on this!

dexterbg, 2022-12-29:
The server issue has been resolved, see: https://community.openchargemap.org/t/service-unavailable-503/740

I'll keep this open as an enhancement request.

dexterbg, 2022-12-29:
Please test, build is in `latest`.

basak, 2022-12-29:
Using 4.1.0 built from `latest`, I see the "Enable OCM" checkbox, and disabling the box hides the charge points. I trust that this means it's not actually fetching them, but with the service API outage resolved, I'm of course no longer getting errors so I can't actually verify that easily directly.

I think this can be treated as resolved then. Thank you for the fast work!

dexterbg, 2022-12-29:
You could check the log or add a firewall rule to be sure, but I think I've added guards to all API calls.

closed 2022-12-29 by dexterbg