openvehicles/Open-Vehicle-Android #146: Notifications don't work on Android 13

billybednar, 2022-12-29:
Starting with Android 13, notifications are disabled by default and the [POST_NOTIFICATIONS](https://developer.android.com/develop/ui/views/notifications/notification-permission) runtime permission must be granted for notifications to work. This app targets Android 13 (as of c23b3dbd772bdae7e82601797cd3a71c2623a78c) and does not request this permission, so notifications do not work. As the [permission is not declared](https://github.com/openvehicles/Open-Vehicle-Android/blob/72fae9f4cf431c04796aa0c24ed9830f141e25ca/OpenVehicleApp/src/main/AndroidManifest.xml#L12-L23), it can not be manually granted from app settings nor by using ADB.

dexterbg, 2022-12-30:
I can only test the fix in an emulator, can you please try on your Android 13 device?

billybednar, 2023-01-03:
That fixed it.

The behavior when the notification permission is denied doesn't seem ideal, but I'll make a separate issue for that once I play with it a bit more.

closed 2023-01-03 by billybednar

markwj, 2023-01-09:
FYI: A user 'Larry' provides the following link regarding this issue:

https://source.android.com/docs/core/display/notification-perm

dexterbg, 2023-01-09:
I already published the fix on Google Play yesterday, maybe 'Larry' can test that?