openvehicles/Open-Vehicle-Android #147: Getting a different car in app

yusijs, 2023-01-10:
I'm honestly not sure what is happening, but suddenly the app started showing me the wrong car. Not sure if it's real or not, but I'm able to send commands etc as well which seems a bit sketchy. 

 [image]

[image]

dexterbg, 2023-01-10:
This is a server (perl) issue, not App related. See https://github.com/openvehicles/Open-Vehicle-Server/issues/2

On which server are you, and what is your vehicle ID?

yusijs, 2023-01-10:
Apologies - I got the correct information via the api, so assumed it was something funky with the app. I'm on dexter-web.de, and the vehicle ID is `ronnielau`

dexterbg, 2023-01-10:
I've installed an update, please contact me if this happens again.

closed 2023-01-10 by dexterbg