openvehicles/Open-Vehicle-Android #148: saving PIN for lock/unlock

breed, 2023-01-16:
for the tesla roadster, it would be nice if the PIN could be saved in settings so that you don't get prompted every time.

i would guess the PIN would be added as one of the Parameters in the setup. right?

markwj, 2023-01-17:
I would be concerned about this from a security point of view (especially on North American roadsters without the immobiliser). The PIN acts as a dual factor, as it is not stored anywhere. If we store it in the App / Car Module then that seems problematic.

breed, 2023-01-17:
the pin is such a weak factor. it is easily bruteforceable. it would still be protected by the phone security. if people want to be prompted everytime for the PIN, they could leave it unset.

Picard0147, 2024-01-09:
Saving pin in app mean that no one can see me when i punchinch it in, so when phone is secure (no malware loaded) then it could be safer than being forced to enter pin every time i want to use pin protected funtion.
Moreover saving pin could be step toward one nice to have feature - biometric protected Lock/Unlock widgets.