openvehicles/Open-Vehicle-Android #149: Missing images Kia Niro EV (MY2023)

andrejs2, 2023-01-29:
Hi all!

Is there any chance to get images for new Kia Niro EV (MY2023)?

Thanks!
Regards,
Andrej

glynhudson, 2023-01-30:
Does OVMS work on the new Kia Niro 2023? 

Here's an example of a pull request to add images, take a look at the files changed it's not too difficult to do yourself: https://github.com/openvehicles/Open-Vehicle-Android/pull/124/

andrejs2, 2023-01-30:
> Does OVMS work on the new Kia Niro 2023?
> 
> Here's an example of a pull request to add images, take a look at the files changed it's not too difficult to do yourself: #124

Thanks Glyn for the reply and link. I'll certainly try to add images...

About new Niro...it works, not the same as e-Niro, but I get quite a lot of data (metrics). Sending commands don't work (lock, unlock, heating/cooling....), status of locks is reversed, open door indicator works. Tyre pressure is wrong... and the BMS cell monitory doesn't work either.

dexterbg, 2023-02-03:
We've also got a Wiki page on this: https://github.com/openvehicles/Open-Vehicle-Android/wiki/Adding-Vehicle-Images

andrejs2, 2023-02-17:
Spent some time figuring out how to use all these to contribute but I'm not keen in programming nor in any work with github. The only thing I managed to find is an image of a black Kia Niro EV (latest model) and compressed it. 

So I will wait till someone skilled bites into this :(.
Thanks for you help anyway.

closed 2023-02-17 by andrejs2