openvehicles/Open-Vehicle-Android #161: Wheels flipped on 2020 nv200

jonopowell, 2024-02-02:
I recently blew a tyre :D so I was looking at ovms and comparing with the App and the in-car display.
On the incar display I see my rear near side (I am in the UK we driver on the RHS) at 1psi.
When I issued the command to ovms via the web, it returns Rear Left at 1 psi.
The Android app however shows the rear off side at 1 psi.
[image]
[image]

dexterbg, 2024-02-03:
Please post the output of commands `tpms` & `met lis v.t.`.

jonopowell, 2024-02-05:
OVMS# tpms
TPMS status              FL       FR       RL       RR
Pressure...[kPa]:     313.7    313.7    315.4    289.6  [stale] 
OVMS# met lis v.t.
v.t.alert
v.t.health
v.t.pressure                             313.711,313.711,315.435,289.58kPa
v.t.temp

the new tyre is at 42psi because I haven't had time to change that yet, the others are all at 45psi. The lower pressure tyre is definitely physically the rear near side (i.e. Rear Left), not the rear right (so I guess the issue is OVMS not the  Android app!)

dexterbg, 2024-02-11:
Thanks for the clarification, that's what I expected.

There is no special handling in the code for right hand sides vehicles, probably the wheel layout needs to be mirrored in that case -- and of course the OVMS needs to either detect RHS vehicles or get a config for that.

Please contact the Nissan maintainer on this.

closed 2024-02-11 by dexterbg