openvehicles/Open-Vehicle-Android #164: Readability issues on "battery" page

malyjaponec, 2024-03-07:
I have just some cosmetical suggestions what to do on "battery" page.

1. white text on green battery is not readable very well. I suggest to change green color to darker or make text black when there is geen under it and white when battery is empty. I am not sure what to do when geen end crosses the number.  Maybe the black shadow or black glow around white text can also solve this issue.

2. The information about charge is very tiny next to the connector. Please bigger. The connector might be just symbol on the side but the  voltage and current should be way bigger. Alocate half of the screen with car image and make text size 3x5 pixels does not make sense. 

3. Missing big number for power when charging. The amperes or km/h (please do not use kph ...this is not correct, k is not km it is just kilo that means 1000 times shortin km to k is really bad habit, please use km/h or km.h-1 or if you insist on "per" kmph) is not what most people here want to see. It is nice to se how much km/h it puts into car but most people things in therm of Watts. They want to see "charging speed: 75kW"

4. landscape mode. Have you ever tired to use app in landscape? If the car is put side of the battery symbol it can work but not like it is now that whole stuff is made even smaller on 1/3 screen space and 2/3 are empty.

And this is only basic problems that i see in this page, for me who needs glasses for reading as getting old using small tiny is  starting to be really. I also use app to see SOC during driving because car does not show it at all and  having that phone horizontally because other apps is no way.

dexterbg, 2024-03-17:
The layout is all quite old legacy, you're welcome to rework this. A graphic designer has contacted me and may be able to help you. But I suggest waiting for the Kotlin migration to finish, as the UI classes currently are still Java.

markus-kraemer, 2024-03-20:
The graphic designer was me. If i could help you after the Kotin migration... i'm here :)

I like the product but the layout is really quite old legacy. So maby i can hel a liitle bit.

[image]
[image]

malyjaponec, 2024-03-20:
Nice. This looks much better than old UI.

askarsyzdykov, 2024-08-01:
Hi @markus-kraemer!

I'm an Android developer. If you have a Figma project with a design, I can implement it by the end of August.

developerfromjokela, 2024-11-30:
> The graphic designer was me. If i could help you after the Kotin migration... i'm here :)
> 
> I like the product but the layout is really quite old legacy. So maby i can help a liitle bit.
> 
> ![01](https://private-user-images.githubusercontent.com/12045730/314686074-3f5e165c-9e92-4178-b4a5-e43d4115be96.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTg1MjIsIm5iZiI6MTczMjk5ODIyMiwicGF0aCI6Ii8xMjA0NTczMC8zMTQ2ODYwNzQtM2Y1ZTE2NWMtOWU5Mi00MTc4LWI0YTUtZTQzZDQxMTViZTk2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDIwMjM0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzMjdlNGRhN2ExZWY4NjViMDhhNmQzMmNhNTJmM2E0N2ZhODlmODMzNGE3OWJjNmMwMWFhZjEwZGM1Yzg2NzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gId8igm7uatB8IzBZO0e-whW1Gj4rujnL0hctmXhFW0) ![02](https://private-user-images.githubusercontent.com/12045730/314686109-c6c64ae0-2b3c-4372-b6b5-8f70ce510c88.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTg1MjIsIm5iZiI6MTczMjk5ODIyMiwicGF0aCI6Ii8xMjA0NTczMC8zMTQ2ODYxMDktYzZjNjRhZTAtMmIzYy00MzcyLWI2YjUtOGY3MGNlNTEwYzg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDIwMjM0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQzYzlmMDUyNTc3M2IzYTA0OGQ4M2U4MGY0M2M2MzQyMDFiZWQ3OWYyYmQ3NmZhNTdjYmE5ZmE5ODI5OTA2MzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.BColNnp1wkAF_7ucQOuE8GekjgonJwdEGHyTgb2SzIs)

ehhh, too much creamy. What about sticking with Material 3 or something simpler?

markus-kraemer, 2024-12-03:
Hello @askarsyzdykov

I'm so sorry... i was under heavy workload the last months. If you still want a facelift, I can port the layout to you as a Figma project and release it. I still think the app is really cool and would be very happy to work with you.

Please excuse my late reply and my bad English :)

askarsyzdykov, 2024-12-03:
@markus-kraemer let's do it!

markus-kraemer, 2024-12-03:
> ehhh, too much creamy. What about sticking with Material 3 or something simpler?

@developerfromjokela 
Thats's cool, too. If you can port the app to a simple material design project with basic components, that would be very future-oriented – because it's basic. Nice too. 

I would also be happy to help you find suitable solutions.

markus-kraemer, 2024-12-03:
https://m3.material.io/components

markus-kraemer, 2024-12-03:
> @markus-kraemer let's do it!

@askarsyzdykov My Figma Skills have to be updated immediately:)

Long time ago... so should i invite you to my noob-file?

[image]

developerfromjokela, 2024-12-04:
> > @markus-kraemer let's do it!
> 
> @askarsyzdykov My Figma Skills have to be updated immediately :)
> 
> Long time ago... so should i invite you to my noob-file?
> 
> A QAD non-colored sketch of the Material Design adaption.
> 
> [image]

Idk something tickles my eyes with this 😜 I'll try doing a sample or something. See what you guys think about that

developerfromjokela, 2024-12-04:
I had an idea that would be cool, if we could make 3d render of the car (optional, if user wants and model is available). Could turn out to be so cool, just like in Tesla app

markus-kraemer, 2024-12-04:
> > > @markus-kraemer let's do it!
> > 
> > 
> > @askarsyzdykov My Figma Skills have to be updated immediately :)
> > Long time ago... so should i invite you to my noob-file?
> > A QAD non-colored sketch of the Material Design adaption.
> > ![Android Compact - 1](https://private-user-images.githubusercontent.com/12045730/391912035-4c2e6ca4-7c50-4358-9d7a-f66675d8622d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMzMTA4NzMsIm5iZiI6MTczMzMxMDU3MywicGF0aCI6Ii8xMjA0NTczMC8zOTE5MTIwMzUtNGMyZTZjYTQtN2M1MC00MzU4LTlkN2EtZjY2Njc1ZDg2MjJkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA0VDExMDkzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgwMGJkNDM3YTUxOTUzMWE1MDY2ZDMxNDk4MDU5NjA3NzU5YzIxMGExZmM0YzUzN2JhZTVlOTI5Y2E4OGUwY2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8pfoVknVHd_UfeIP0-OGggJOSF8_dzBzBcRw-Xzmn0A)
> 
> Idk something tickles my eyes with this 😜 I'll try doing a sample or something. See what you guys think about that

@askarsyzdykov I'm faster in Sketch :)

[image]

developerfromjokela, 2024-12-04:
I meant that all elements of design use Material 3, not only bottomnav and
actionbar. There's really no better app to take inspiration from.than Tesla
:) something like that, simple yet effective

markus-kraemer, 2024-12-05:
Typo, Icons, Sizes... a great Starting Point for the Demo 👍🏻

[image]

developerfromjokela, 2024-12-25:
> Typo, Icons, Sizes... a great Starting Point for the Demo 👍🏻
> 
> ![Info](https://private-user-images.githubusercontent.com/12045730/392711385-b9fe342a-2dcc-4f9a-b1b8-a4126c188e6f.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxMzM2MzQsIm5iZiI6MTczNTEzMzMzNCwicGF0aCI6Ii8xMjA0NTczMC8zOTI3MTEzODUtYjlmZTM0MmEtMmRjYy00ZjlhLWIxYjgtYTQxMjZjMTg4ZTZmLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNVQxMzI4NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YWUwODQ1NDlmNTQzNmNmNDc3ODhiN2IyNzUwMmMzOGNkYTFjZjJlMTc2OWYwZmM1Y2IyNDg2ZjA4NzJiNjM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xWujvTinpZT65GSHdqYUJm4-9eX2zdGajXEek7mFc28)

[image]

developerfromjokela, 2024-12-25:
Those smaller buttons under charging card can be customised for custom commands, rearrangement etc. Each button handles commands separately so customisation is out of the box

developerfromjokela, 2024-12-30:
Check out [https://github.com/openvehicles/Open-Vehicle-Android/tree/test-dfj-ui/latest](https://github.com/openvehicles/Open-Vehicle-Android/tree/test-dfj-ui/latest)!

markus-kraemer, 2025-01-09:
A great step in the right direction 👍🏻

Thank you @developerfromjokela