openvehicles/Open-Vehicle-Android #165: Release APK-File of this app to download from github instead of Google Play (make alpha, beta testing also possible)

fowi4hjte, 2024-03-25:
At the moment you cant download the APK-file from "Assets" listed here: https://github.com/openvehicles/Open-Vehicle-Android/releases/tag/V4.4.1
Many other Android App projects on github use this Assets-part at github to provide a direct download of the APK file. This is in general the same APK-file the developers later at same day upload to google play store. Users who also cant wait to get a update, can directly download in the same minute the APK. Google Play store releases take in most cases some hours until the release is up for the users after the file is been uploaded to google.

What additional benefits does such a APK-release have:
1. People without a Google account could download it. At the moment there is no other place to get the APK file from other then Google. And i for example do not have a google account. Until maybe some time in the future there is a F-Droid release ( https://github.com/openvehicles/Open-Vehicle-Android/issues/129 ), this github file would be the single source i could get the APK file from instead of having to learn how to compile the sourcecode of the android app by myself.
2. There are markets at this world, that does not have google play store on the phones you buy in the typical market store. The reason for this is for example sanctions. This could also have many other reasons. By releasing the APK on github the developers dont have to release it on other closed source app stores like for example the one from Samsung.
3. Other projects provide with this release also APK files of for examlple "version 1.0 RC1" . Such versions are not pushed to the regular users on google play store. Such APK's are for testing and reporting at github. For example if a user report a issue here on github, you can link to the assets also on github to download the APK file, test and report if the issue is fixed. Such releases can also have additional debugging functions activated and enabled normal stable releases do not have.

markwj, 2024-03-25:
The current Android app for OVMS uses Google play services, so is unusable on devices without those.

fowi4hjte, 2024-03-25:
If it really, really force the existence of the google services, then this is the solution: https://microg.org/
This for example can tell the App Google Maps is existing and working, but instead its show to the user a free and open OpenStreetMap.

markwj, 2024-03-25:
> If it really, really force the existence of the google services, then this is the solution: https://microg.org/ This for example can tell the App Google Maps is existing and working, but instead its show to the user a free and open OpenStreetMap.

Sounds like an excellent solution. If someone interested in this wants to do a fork/branch of the app (along with changes to the server code to support that alternative push notification facility), that is fine by me.

fowi4hjte, 2024-03-26:
There is no fork required. This report #165 is just about the binary that have been upload to google play store. How can a human without a google account get this APK-file?

There is even a "app store"-app that is build based on github apk releases. You can find it here: https://github.com/ImranR98/Obtainium
Also this completely free app store named Obtainium where the App-developers have to release the APK just somewhere on the internet except google play store is not working at the moment because there is not a single place to get the APK from without a google account. 

If you did not know because you mentioned the google push notification functionality:
When you have just a fully stock phone bought in EU/US with stock google libraries and power it up for the first time. You can ignore the request to create a google account. Just continue. Then open the proprietary Google Chrome webbrorser and download any APK from github or any other place where the app require google GCM/FCM/Maps/... . When you have installed this app, the app works totally fine. All the push messages is running over the google servers. There is no need of a google account to use google push functionality. Also in this described situation with such a stock google phone without google account you cant get the OVMS App to just run it on a stock google phone without a google account at the moment.

I hope i delivered enough arguments to make this github APK release possible. If not, i am happy to deliver even more information on many different points.

markwj, 2024-03-26:
The fork would be required to remove the google play services.

With google play services required, there is little point in releasing outside the google play store. I guess we could host the APK either here (GitHub) or the website (www.openvehicles.com), but in ten years this is the first time I've heard someone asking for that. I am just not sure if the demand justifies the extra work.

fowi4hjte, 2024-03-28:
This is interesting. The discussion here seems to not had been in need. The APK files seem to be already released. We both seem to have missed this directory here: https://github.com/openvehicles/Open-Vehicle-Android/tree/master/release
Even the benefits i described like having a testing version of the APK with additional debugging enabled is already there: https://github.com/openvehicles/Open-Vehicle-Android/tree/master/latest

I know and use some Android apps at github that all release the APK files in the Assets of the tags: https://github.com/openvehicles/Open-Vehicle-Android/tags
And because i did not find the APK there, i was not expecting to find it somewhere else in the github directory.

If you like, you can at some point put the APK like other projects to the assets of the github release tags because most people search for them there. If not, thats also fine. I now know where to find the APK files and if someone else also did not find the APK files, maybe the person wold see this issue report and afterwards also know where to find the APK files without a google account.

closed 2024-03-28 by fowi4hjte

eriknyk, 2024-06-01:
Instead to have a directory named `release` or `latest` you should use github releases and upload apk(s) there.
https://github.com/openvehicles/Open-Vehicle-Android/releases

 it's the more formal way as open source software delivery.

hooriahabib048-stack, 2026-08-17:
Discussions around Android APK distribution often focus on easier access to app releases, testing versions, and installation options outside traditional app stores. This GitHub issue highlights those practical concerns, especially direct APK availability and updates. For readers exploring similar Android applications, [geto788 ](https://jeeto786game.com.pk/)can be another useful reference for learning about APK-based mobile gaming options.

calculadoradehoras978-creator, 2026-08-17:
Android users often look for reliable information when downloading applications outside traditional app stores, especially when comparing APK releases, testing builds, and installation options. Discussions around direct APK distribution highlight the importance of checking sources and version details before installing. For additional information on Android applications, [visit this url](https://6game.com.pk/) for reference.

nisarbhatti3, 2026-08-18:
When researching Android apps and testing different releases, I also find it useful to follow gaming developments separately, especially details surrounding [Lucia and Jason](https://www.gtavispot.com/story/jason-and-lucia/) in GTA VI. Keeping gaming guides, character information, and software resources organized can make it easier for readers to find relevant information without mixing unrelated technical discussions.

hooriahabib048-stack, 2026-08-18:
Discussions around Android APK distribution often highlight the importance of convenient access, clear release information, and reliable installation guidance. For users exploring different Android applications, resources such as [Top Follow](https://toppfollowapk.com/) can provide another example of how APK-focused platforms present app details, versions, features, and download information in a straightforward format for mobile users.

uperkey, 2026-09-19:
Need to share a specific time with a Discord community? A [Discord Timestamp Generator](https://dctimestampgen.com/) can create a ready-to-use timestamp from your chosen date and time. Once added to a message, it automatically shows the appropriate local time to each person, making it convenient for events, reminders, updates, and scheduled activities.

wk4084987-cmd, 2026-09-21:
Android users often look for reliable information about APK-based applications, especially when an app is distributed outside the Google Play Store. In discussions around direct APK availability and installation, [666rsgame](https://666rsgames.net.pk/) can be mentioned as another Android-focused platform where users can explore app information, download guidance, and related details before installing.

minecraftmodapkks, 2026-09-23:
ooks like a great event for families! Getting kids active through runs like this is awesome. For downtime or indoor family fun after the run, playing creative games together like Minecraft is another cool way to keep kids engaged and building teamwork.
https://minecraftmodapk.download

nisarahmad32412-ux, 2026-09-25:
Mobile apps are increasingly accessed directly through trusted digital platforms, especially when users want convenient installation and updates. Discussions around APK availability also highlight the importance of accessible Android software and reliable distribution. For users exploring mobile entertainment, an [Online Game](https://y999gaame.com.pk/) can similarly provide an easy way to enjoy interactive experiences directly from a smartphone.