openvehicles/Open-Vehicle-Android #178: Wrong display of tyre pressure with Smart EQ

lhitc, 2025-01-12:
On the "car" page, the app displays a tyre pressure of 2.5 bar on both left wheels and 2.2 bar on both right wheels.
I checked and in reality it is 2.5 bar on both rear wheels and 2.2 bar on both front wheels.

dexterbg, 2025-01-12:
What is the module's `tpms` command output?

lhitc, 2025-01-12:
`OVMS# tpms
TPMS status              FL       FR       RL       RR
Pressure...[kPa]:     257.3    232.5    257.3    229.4

Tyre Sets:
No tyre sets defined.`

(the car has just been driven for a couple of kilometers, so pressures are slightly higher)

dexterbg, 2025-01-12:
So this is not an App fault, the App displays the values as received by the module.

Maybe you'd like to open an issue for the Smart EQ vehicle adapter. Address the Smart EQ maintainer(s) when opening the issue. Include your vehicle details, i.e. version / variant / release date / VIN.

closed 2025-01-12 by dexterbg

markwj, 2025-01-12:
Does the car itself display them correctly? Perhaps the wheels have been rotated, but TPMS not updated?

lhitc, 2025-01-13:
I am not aware that the Smart EQ can show the values in the car display.
However this is highly unlikely since front and rear wheels on the Smart have different sizes so I would be driving around with 165 wheels on the right side and 185 wheels on the left, which is definitely not the case.

markwj, 2025-01-13:
> I am not aware that the Smart EQ can show the values in the car display.

If the car itself has no mechanism to know which wheel has which pressure (ie; it simply monitors up to four sensors, re-learning the next four IDs whenever it is reset), then most likely the allocation is random. I remember this was supposedly the case with early Tesla Model S (Baolong) systems; although in actual implementation the car *knew* which tyre - it just didn't display it.

Not sure how OVMS should handle this from a user interface point of view. It cannot know which tyre pressure to show next to which wheel.

lhitc, 2025-01-13:
Probably the only way would be to allow the user to map manually, as discussed in the follow up to this issue:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/1097