openvehicles/Open-Vehicle-Android #180: New UI feedback

glynhudson, 2025-02-07:
As I mentioned, I really *love* the new UI, I've been using it for the past week or two. It's a massive upgrade in almost every way, I love the  vehicle animations and the new modern look is awesome. 

Here are a few though on the UI, all of this is my personal opinion, so feel free to ignore :-) 

- SoC display is small: the most important metric and the reason most people use OVMS is to check the SOC. On the old app the SOC display and battery icon was very clear. On the new app it's rather small! 

[image]

- Quick actions: it's not clear from the icons what each quick action button does, it's not ideal to have to press them to find out! Also most of the quick actions are not supported on any vehicle apart from Roadster. Maybe the quick actions could be customised so users could only show the ones they want? Personally, I would like a remote climate quick action, most of the other quick actions are not supported by the Leaf or most other vehicles. 

- Control page: Would it be possible to label the buttons to make it clear what each button does? This is quite important for lock/unlock, since I'n sure many users will assume that tapping the padlock will lock the car rather than unlock it! Again, it would be useful to be able to hide functions which a particular vehicle doesn't support. I guess ideally, each vehicle integration would present a list of which features are supported then the app could automatically hide unsupported functions. 

[image]

- Missing parking duration - on the old app it was possible to see how long the vehicle has been parked for, this seems to be missing from the new UI?

[image]

- Show SoC on the vehicle switcher page - would it be possible to show the Soc of each vehicle on the vehicle switcher page? This would be really cool :-) 

[image]

dexterbg, 2025-02-07:
Quick actions are customizable already, try a long click on the button area.

glynhudson, 2025-02-07:
Nice! That's fantastic. Feel silly to have missed this.

When the customise draw is opened the icons are labelled which is nice

dexterbg, 2025-07-22:
Settings → Global Options (icon upper right) → Old UI

developerfromjokela, 2025-07-22:
> I absolutely HATE HATE HATE the new design.
> It's crap. 
> Easy to find information is hidden. It's too easy to accidentally click something.
> What a terrible decision?
> 
> How to downgrade? And where the F do I find all the settings to set up the old app again.
> This is such a gigantic waste of everyone's time. An insult.

What easy to find info you need? Suggestions are open 😄

developerfromjokela, 2025-07-27:
> As I mentioned, I really _love_ the new UI, I've been using it for the past week or two. It's a massive upgrade in almost every way, I love the vehicle animations and the new modern look is [...]

I'll tackle some of these issues if not done already, finally have some time on my hands. any other suggestions are welcome, throw them in the comments.

dexterbg, 2025-07-27:
> Can't quite find more than one review just stating that they don't like the update :) If you see more reviews in play console won't mind if you attach some of them here thanks.

Most were in German language, you can see them in the Play store after changing your preferred language.

moffetx, 2025-07-29:
I've also noticed the top down view (Controls page) no longer shows when a door is open on my eNV200.

JeffAnton, 2025-08-02:
I'm very unhappy with the new UI. And it's not just small problems, much of the style is poor.
To be fair, I'm a Roadster driver and the previous UI was probably too focused on that one car which is now very rare.
I also have tried the new app for a while now, trying to get used to it, to give it a fair chance. But I'm still frustrated.

I measure a good UI by how >> few << interactions are needed to get to the information I want. The new app fails on this.

I recommend whoever designed this to get and read the books by Edward Tufte. In particular, _Envisioning Information_.

The old UI would move between screens by swiping which is very quick. The new one you have to search through and read buttons to decide which one to press. That's a lot more cognitive work. Also, to move from one screen to another you must go back to the home screen and choose the next one. I want a complete information about my vehicle and when I used to swipe quickly, now it's many clicks which involve careful location of the back button and the next screen buttons.

The buttons and fields also use a lot of decoration in the form of boxes with rounded corners. This might be to highlight the buttons, but when everything is highlighted, nothing is really highlighted.

The "Energy" screen puts a rounded box around each item. This is more "ink" and lines which are not needed. See Tufte to simplify this. If you want separation of items to make them clearer, remove the boxes and make a subtle background change on alternating lines. This will allow more dense data display while maintaining readability.
Some items on the "Energy" screen do not IMO belong there. "Last Trip" and "Drive Mode" do not belong there.

The earlier UI having a vehicle image with numbers overlayed and around the image was a very good module and that seems pretty much gone. Like the "Climate control" has ambient and cabin in a box at the bottom while the information would be better overlayed on top and beside the vehicle image.

The "Controls" screen could IMO have the climate and some energy information all overlayed. Maybe you need an "Overview" screen which would show as much as possible. It seems strange the buttons on the "controls" screen are redundant with the home screen. Seems like with some careful design, these screens could be merged.

Anyhow, I've only just highlighted now the most annoying issues I have with the new UI. I'm glad someone is working on this, but this needed input along the way and IMO a gradual improvement rather than a big change which will likely need substantial changes.

dexterbg, 2025-08-02:
Thank you for the detailed review and reasonable criticism. We will try to incorporate some of your wishes in the further UI development.

Swiping between the screens was broken for quite a while now with the map fragment blocking the swipe action (probably may be solved somehow though), and -- more problematic -- interfering with the new Android swiping gesture standard (swipe from the left/right edge = navigate back). It was still possible to swipe, but you had to take care where exactly you put your finger down, I often accidentally triggered the wrong action.

I'm not sure if refitting swipe navigation is an option, it may have major conflicts with the new NavHost/NavController style navigation. Both seem to have separate navigation stacks. There are third party libs that try to integrate both, but those often come with their own issues.

developerfromjokela, 2025-08-02:
> I'm very unhappy with the new UI. And it's not just small problems, much of the style is poor. To be fair, I'm a Roadster driver and the previous UI was probably too focused on that one car which is [...]

1. The design is a standard off-the-shelf Material 3 that is common across whole android nowadays.
2. I don't know edward tufte but there isn't a lot of vehicle data to begin with, at least on leaf and others. Most necessary ones range, soc and overall state of the car is shown on front. Quick actions are also rather easily accessible. I don't know what amount of roadster data was showin in the old UI but for 99% of non-roadster cars there was no data that would have a common usage on daily basis, like inverter temp. I'm sorry to phrase it like this but let's face it, it's data for the nerds to have a look at for fun once a week ;)
3. Could you also tell what data you're accessing quite often from other menus? I get away most of the time by staying on the home screen. Climate data is shown underneath the navigation button and so is the location. 
4. the tabs controls and climate leave place to have more controls in the future, for example adjusting temperature, etc. 
5. and afterall maybe you've used the old UI so long can't let it go? ;) I've used the old UI for under a month and happily use the new one.

JeffAnton, 2025-08-02:
1. "Off-the-shelf" It may be common but that does not make it good. Frankly, a lot of UI is poor but designers just follow the guidelines from the mega corps who have their own issues which are often counter to good design just in the cause of differentiation. I.e. because everyone else does it that way does not make it good. It makes it familiar.  And I don't think you are using these android features correctly. For instance when I look at notifications on my android, many items are in a group box with outer rounded corners and thin boundaries between items. Seems like you're separating things with major separators not minor one. And that wastes a lot of space and is visually poor.
2. Vehicle data - The Roadster may have more data, but that is a good thing. That other cars have less should not diminish my experience. Different vehicles will have different data, you should be able to adapt to that and not use the minimal common data. This is something you could evolve to. But, your language is insulting. I think you do not care about quality. Car people are often nerdy about their cars and they want all the information they can get. If most people just don't care, and that who you are targeting, IMO you're dooming yourself to have a product which will have no value beyond what the car manufacturers will produce and support. If you do no better than that,  you are dooming your project. For the Roadster and probably for other battery cars, the battery temperature is critical. After every long drive I want to see the battery temp so I can decide if the car needs to cool down via starting a charge right away.
3. Data: I look at a lot, tire information - temp and pressure  Battery temp. PEM/inverter temp. I look at these all the time. After almost every drive. I want it all on one screen with items in positions where I can easily tell which number is which. "Climate" is of secondary interest to me, I can tell if it is hot outside on my own. Again, maybe most people will want that, but again, you're dooming yourself if you don't differentiate OVMS beyond what the basic stuff is. Other data I look at weekly is stuff like CAC and SOH (for non-roadster people Calculated amp-hour capacity and "state of health" of the battery.) Data usage by the app is another thing I look at occasionally which I have not found at all. A good UI will steer people to this seldom looked at data, but that steerage isn't working for me at all.
4. Sure - you need stuff for the future.
5. Again, this seems insulting. I had an app which suited me. It could have been improved with a new UI, but really it's not better either in usage or functionality.

developerfromjokela, 2025-08-02:
> 1. "Off-the-shelf" It may be common but that does not make it good. Frankly, a lot of UI is poor but designers just follow the guidelines from the mega corps who have their own issues which are [...]

Can you send a screenshot and highlight which element you're talking about on your first point? 

Just like the old app used base android design, the new one does. The difference is that old design was made in 2010 and now it's 2025 :) times have changed and to keep it consistent with most of other Android apps and more importantly with the android itself. On this note the app icon would need a vectorization also to keep up with the standards.

Well I agree with it's less shown data for keeping it clean for the average consumer as they say. I decided to not do a data overload because there was only few data points for most of ovms users.

I'm not trying to insult anyone, but data overload isn't great either. It can get quite complex quickly. Current issue with ovms is that there's no way to know vehicle's capabilities to adapt the UI better. Yes we could hard code if statements all over the app (it's been done to some extent) but that ain't perfect when vehicle catalog is also growing. There has been some talks to implement this on server side, so maybe do it properly than get quick results and have to rewrite it soon again? :)

It's not like I don't want to see the data, but I don't want to see it every time I open the app. I probably look at it only couple times a week. I find it acceptable to have to click at least one menu button to see more of the technical stuff, don't you agree? Most of what I do on the app is controlling charging, sometimes checking on location and checking soc/range. Twizy seems to be also one of those vehicles with exceptional amount of data, not only roadsters.

Also about CAC, hasn't it always been available only from some menu in vehicle list? It was never shown on front page even on old UI? It was in the settings where list of vehicles is. Please correct me if I'm wrong. SOH is widely shown in charging tab, also battery temp. I guess we could add it somewhere there.

But about all the juicy data, maybe we could implement a mode in the app to show all additional data to suit your needs while not enforcing it on every user. A checkbox in settings for example.

But again, ovms app as the saying goes needs to walk first before running. The way it was previously and especially when you look at current iOS it's borderline unusable. Now that it has been brought up from the ashes we can improve it and for instance moving obd voltages, data usage charts to a submenu rather than being under deep in the settings menu. I also like to look at the data now and then but not every time. 
There's still a lot to be done, as a LEAF user I'd like to see battery cell information, mv differences and weak cells like in the web UI but afaik it's not possible to request this info from server, TBD.

dexterbg, 2025-08-03:
@developerfromjokela On the "Unknown vehicle type" issue: you assumed the car type is a fixed code, that's not the case. It's a dynamic metric that in some cases includes additional chars/digits based on the model year or variant. The roadster is one of the vehicles that do that. You can test for the known vehicle type prefix (that's fixed), and if the type is unknown it better could display "Vehicle type $car_type" instead of "Unknown".

I prefer the large home screen buttons over the menu style, they provide better visibility and a clear indication of the functional areas along with their respective most important state info. Regarding the single metric displays on the energy screen, I second they could be displayed in a more compact layout (e.g. striped table style), also providing a better distinction between value displays and buttons.