openvehicles/Open-Vehicle-Android #181: New UI Bug: crash while when rapid charging

glynhudson, 2025-02-09:
When rapid charging the app crashes immediately after opening.

As soon as rapid charging session is stopped the app returns to working as normal. See video

dexterbg, 2025-02-09:
Trying to emulate a Leaf fast charging session with the DEMO vehicle, but I cannot get the App to crash. It even displays another plug for the chademo charge type.

Do you have the server data export of that session, would you mind mailing me that? I should only need the "S" & "D" tables.

glynhudson, 2025-03-10:
Sorry for the delay, here's the S & D tables which include a couple of rapid charging sessions today: 

[ovms-demo-250310.zip](https://github.com/user-attachments/files/19166313/ovms-demo-250310.zip)

dexterbg, 2025-03-30:
Glyn, I finally got the time to dig into this. Build 5.0.7 should fix this, please test.

glynhudson, 2025-03-31:
Fantastic, I'll test it. Is the latest build apk available from somewhere? Or do I need to build it?

glynhudson, 2025-04-13:
Nice work! I can confirm this issue has been fixed 👍

closed 2025-04-13 by glynhudson