openvehicles/Open-Vehicle-Android #187: Map location not working Build 5.0.7

glynhudson, 2025-04-02:
I'm testing V5.0.7, I've noticed the map location is not working. The text location shows correctly on the main page, but opening up the map view results in a blanks screen. 

I'm running Android 15 on Google Pixel 7

[image]

dexterbg, 2025-04-03:
I can confirm something is out of order. My phone still displays the map, but detailed only for the region cached. When moving the map view, other regions are shown only with very coarse road outlines.

There was no change to either the map integration or the API key though… this looks more like a maps server issue.

dexterbg, 2025-04-03:
Or we've missed another Google deprecation? https://issuetracker.google.com/issues/404999856?pli=1

dexterbg, 2025-04-03:
Franco updated the maps lib to 18.2 in December, and the map still works in the emulators. This is a device specific issue, and this is the culprit:

https://developers.google.com/maps/documentation/android-sdk/renderer#supported-devices
> To use the upgraded map renderer, devices must meet these criteria:
> · Using Google Play services version 21.39.14 or later

I checked for an available Play services update, and there was one, without any notification before. I did the update, and voila, maps working again.

Please check your device for a missed Play services update.

glynhudson, 2025-04-12:
> Do you build your own apk or did you install the apk from the [latest](https://github.com/openvehicles/Open-Vehicle-Android/tree/master/latest)?

I built the APK myself. I've now switched to running the pre built latest and it's working now. Sorry for the hassle.

closed 2025-04-12 by glynhudson