openvehicles/Open-Vehicle-Android #208: 5.1.2 Android App crashes

markwj, 2025-07-28:
I've had a few messages in to support@openvehicles.com regarding crashes in the 'latest' Android App on startup. No much supporting information provided except one user saying it is v5.1.2 and all saying the crash is on startup.

Example messages:

> The app closes immediately after opening, making it impossible to use. This happened shortly after the update message started appearing. I should add that I already had a new skin, which also installed itself some time ago. I have a Samsung S24 Ultra, so the app should work. Now it doesn't. What am I have to do? Jakub Dąbrowski

> Seit heute funktioniert die ovms app auf einem Samsung a55 leider überhaupt nicht mehr. MfG. Gerhard Fakler.

> Ich korrigiere mich die App funktioniert überhaupt nicht mehr !!!!!!! EGAL AUF WELCHEM GERÄT !!! Gerhard Fakler.

dexterbg, 2025-07-28:
Those are probably all Smart drivers, the Smart parts in the new UI had some serious issues that never got reported by any beta testers.

Release 5.1.3 should fix hopefully all or most of them and is in review at Google since Friday, should get approved today.

dexterbg, 2025-07-28:
Just checked, 5.1.3 has been approved. I need to wait for the next Android vitals updates to see what issues remain.

markwj, 2025-07-28:
Feedback from Jakub is:

> Good news, It started working by itself. I`ve forgot, I have a Smart 

So I guess he got the update that fixed this, from google play automatically.

closed 2025-09-27 by dexterbg