openvehicles/Open-Vehicle-Android #212: ANR in ui2.pages.HomeFragment.initialiseTabs due to synchronous Geocoder call

dexterbg, 2025-08-03:
@developerfromjokela 

com.openvehicles.OVMS.ui2.pages.HomeFragment.initialiseTabs
Input dispatching timed out
```
"main" tid=1 Timed Waiting
  at jdk.internal.misc.Unsafe.park (Native method)
  at java.util.concurrent.locks.LockSupport.parkNanos (LockSupport.java:269)
  at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquire (AbstractQueuedSynchronizer.java:756)
  at java.util.concurrent.locks.AbstractQueuedSynchronizer.tryAcquireSharedNanos (AbstractQueuedSynchronizer.java:1126)
  at java.util.concurrent.CountDownLatch.await (CountDownLatch.java:276)
  at android.location.Geocoder$SynchronousGeocoder.getResults (Geocoder.java:396)
  at android.location.Geocoder.getFromLocation (Geocoder.java:148)
  at com.openvehicles.OVMS.ui2.pages.HomeFragment.initialiseTabs (HomeFragment.kt:1268)
  at com.openvehicles.OVMS.ui2.pages.HomeFragment.update (HomeFragment.kt:1395)
  at com.openvehicles.OVMS.ui2.pages.HomeFragment.onServiceLoggedIn (HomeFragment.kt:1410)
  at com.openvehicles.OVMS.api.ApiObservable.notifyLoggedIn (ApiObservable.kt:59)
  at com.openvehicles.OVMS.api.ApiService.onLoginComplete (ApiService.kt:617)
  at com.openvehicles.OVMS.api.ApiTask.onProgressUpdate (ApiTask.kt:76)
  at android.os.AsyncTask$InternalHandler.handleMessage (AsyncTask.java:791)
  at android.os.Handler.dispatchMessage (Handler.java:110)
  at android.os.Looper.loopOnce (Looper.java:248)
  at android.os.Looper.loop (Looper.java:338)
  at android.app.ActivityThread.main (ActivityThread.java:9067)
  at java.lang.reflect.Method.invoke (Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:593)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:932)
```

zorgms, 2025-08-07:
There is a bug in the Home Update: if you click on Position and then go back to Home, all menu entries are duplicated.

dexterbg, 2025-08-07:
Confirmed.